lukevp/ESC-POS-.NET #248: Epson USB printer - getting "System.IO.IOException: The handle is invalid"

ddevriend, 2023-11-08:
I am using the newly released USB modules (USBPrinter.cs, DeviceDetails.cs, & DeviceFinder.cs) and 
using "<TargetFramework>net5.0-windows7.0</TargetFramework> in my project files. Evthing seems to complie just fine, but when I run the Console Test, I get the following:

[image]

Any ideas?

ddevriend, 2023-11-09:
I question the "The handle is invalid, " line. Does the "\\?\" need to be formatted differently?

Gary-gr9, 2023-12-01:
I tested this using .net5.0-windows7.0, .net5.0, and .net6.0 with Epson printer but was unable to recreate the issue. Could you please provide me more information about the testing environment.
Did you add these classes to the existing code?
I'll give it a go and try fix the bug if it exists.
Cheers