lukevp/ESC-POS-.NET #251: 2 Worker Services to one thermical printer

detectivejd, 2023-11-17:
Hi, how are you? I wait that you are good, I have the next trouble: 

I´m using 2 worker services implemented in C# .net7 one to ticket impress and other to bills impress but they don´t found of simultane form. 
I show you the images:

ticket impress
[image]

bills impress
[image]

The Impress Code is the next:

Ticket Impress
```
using System.Text;
using ViewParking.Tickets.Raspberry.Entities;
using ViewParking.Tickets.Raspberry.Utilidades;
using Logging = ViewParking.Tickets.Raspberry.Utilidades.Logging;

namespace ViewParking.Tickets.Raspberry.Impresion;

public class EntradaImpresion {
    private string URL_API = "";
    private string RutaArchivoLog = "";
    private bool UsaTicketExtendido = false;
    private long IdClienteTicketEspecial = 0;
    private int TiempoCicloChequeo;
    private EnvioImpresion? prnt;
    //private readonly string carril;
    private List<int> Impresoras = new();
    private bool ImprimoAutomaticos;

    public EntradaImpresion()
    {        
        try
        {
            var configuration = new ConfigurationBuilder()
                .AddJsonFile("appsettings.json", true, true)
                .AddEnvironmentVariables()
                .Build();

            URL_API = configuration.GetValue<string>("URL_API") ?? "";

            var properties = configuration.GetSection("Default");
            RutaArchivoLog = Path.Combine(AppDomain.CurrentDomain.BaseDirectory, "Log");

            var impresorasConf = properties.GetValue<string>("Impresoras") ?? "";
            Impresoras = CargarImpresoras(impresorasConf.Split(','));

            ImprimoAutomaticos = properties.GetValue<bool>("TomaAutomaticos"); 

            TiempoCicloChequeo = properties.GetValue<int>("TiempoCicloServicioMilisegundos");

            UsaTicketExtendido = properties.GetValue<bool>("UtilizaFormatoTicketExtendido");

            if (UsaTicketExtendido)
            {
                if (properties.GetValue<int>("IdClienteTicketEspecial") > 0)
                    IdClienteTicketEspecial = properties.GetValue<int>("IdClienteTicketEspecial");
                else
                    throw new Exception("La configuracion no tiene el identificador de IdCliente para tickets personalizados");
            }

            prnt = new EnvioImpresion(properties.GetValue<string>("Impresora"))
            {
                CantLineasEnBlancoAntesDelCabezal = properties.GetValue<int>("LineasEnBlancoEnCabezal"),
                CantLineasEnBlancoDespuesDelPie = properties.GetValue<int>("LineasEnBlancoAlPie"),
                ActivarNumeracionTickets = properties.GetValue<bool>("ActivarNumeracionTicket")
            };
[... 318 lines cut ...]
                        if (borrado.StatusCode == System.Net.HttpStatusCode.OK)
                        {
                            var esExito = Convert.ToBoolean(borrado.Response);
                            if (esExito)
                            {
                                Logging.Instancia().LogMensajeImpresionFacturas("Se borró la factura " + docCabezal.Numero + docCabezal.Serie);
                            }                            
                        }
                    }
                }
                catch (Exception ex)
                {
                    Logging.Instancia().LogMensajeImpresionFacturas("Error al obtener los tickets para imprimir. Probablemente el WS esté fuera de servicio\nRevisar log de errores :)");
                    Logging.Instancia().LogError(ex, RutaArchivoLog);
                }
            }
        }
    }
}
```

I have tried delete prnt instantation from the constructor and use it in the impress task but it didn´t work.

I need your help with urgency, if you can answer me, I will appreciate it.
Greetings.

detectivejd, 2023-11-20:
Please!, I need to help with this code, The worker services are different applications and they are installed in a raspberry pi as deamon services in Linux

Service to Ticket Impress
[image]

Service to Bill Impress
[image]

I don´t know if on the code have fails or not, I was looking for information in Google but all that I found was useless and I didn´t work it.

Ticket Impress -> Program.cs
```
using ViewParking.Tickets.Raspberry.Impresion;

IHost host = Host.CreateDefaultBuilder(args)
    .UseSystemd()
    .ConfigureServices(services =>
    {
        services.AddSingleton<IHostedService, Worker>();
    })
    .Build();

await host.RunAsync();
```
Ticket Impress -> Worker.cs

```
namespace ViewParking.Tickets.Raspberry.Impresion;

public class Worker : BackgroundService
{
    private readonly ILogger<Worker> _logger;

    public Worker(ILogger<Worker> logger)
    {
        _logger = logger;
    }

    protected override async Task ExecuteAsync(CancellationToken stoppingToken)
    {
        var configuration = new ConfigurationBuilder()
            .AddJsonFile("appsettings.json", true, true)
            .AddEnvironmentVariables()
            .Build();

        var properties = configuration.GetSection("Default");
        var tiempoCicloChequeo = properties.GetValue<int>("TiempoCicloServicioMilisegundos");
        var entrada = new EntradaImpresion();

        Console.Clear();

        while (!stoppingToken.IsCancellationRequested)
        {
            await Task.Delay(tiempoCicloChequeo, stoppingToken);
            entrada.Iniciar();            
        }
    }
}
```

Bill Impress -> Program.cs

```
using ViewParking.Facturas.Raspberry;

IHost host = Host.CreateDefaultBuilder(args)
    .UseSystemd()
    .ConfigureServices(services =>
    {
        services.AddSingleton<IHostedService, Worker>();
    })
    .Build();

await host.RunAsync();
```

Bill Impress -> Worker.cs

```
using ViewParking.Facturas.Raspberry.Impresion;

namespace ViewParking.Facturas.Raspberry
{
    public class Worker : BackgroundService
    {
        private readonly ILogger<Worker> _logger;

        public Worker(ILogger<Worker> logger)
        {
            _logger = logger;
        }

        protected override async Task ExecuteAsync(CancellationToken stoppingToken)
        {
            var configuration = new ConfigurationBuilder()
                .AddJsonFile("appsettings.json", true, true)
                .AddEnvironmentVariables()
                .Build();

            var properties = configuration.GetSection("Default");
            var tiempoCicloChequeo = properties.GetValue<int>("TiempoCicloServicioMilisegundos");
            var entrada = new EntradaImpresion();

            Console.Clear();

            while (!stoppingToken.IsCancellationRequested)
            {
                await Task.Delay(tiempoCicloChequeo, stoppingToken);
                entrada.Iniciar();
            }
        }
    }
}
```

You response me, please!.

igorocampos, 2023-12-05:
Hi, that's too much information, and your code is not even presented with syntax highlight. Very hard to follow, especially with a lot of Spanish embedded in naming and all. 
Anyhow, from what I could understand, you have 1 printer connected but 2 different apps running at the same time wishing to use the printer, but only the first application is able to do so, the second one is throwing an error. Is that it?

If that's the case, you have a race condition between your apps when consuming the printer resource. If you really wish to implement it like that (separate applications using the same printer) you would need a printing spooler, where you would have to manage the readiness of the printer, so there are no conflicts when trying to use it.

closed 2024-09-18 by igorocampos