lukevp/ESC-POS-.NET #267: Interest to support CUSTOM printers

jonekdahl, 2024-02-17:
Hi,

I would like to ask whether there is still interest in adding support for CUSTOM printers to this library. I see there are some previous discussions and attempts (#105, #106) that were never merged.

igorocampos, 2024-02-20:
Yes, definitely. #105 was actually going good, but then it lost traction if you read it through, the last comment says "I will continue to get to the bottom of the barcodes." 
#106 got off track early on. Back then I was not a maintainer of the repo, but that has changed since. So if you want to read all the comments of both PRs and propose a new one (I don't think you could hijack the existing ones). I can certainly review it better now.

Don't forget photos of the end result validating the commands, as I don't own a CUSTOM printer.

jonekdahl, 2024-02-21:
Hi Igor,

Okay, thanks. We have a Custom printer at work and are looking for different software solutions for printing and monitoring. Adapting this library is one option and if we go that route I plan to contribute those changes back.

closed 2024-02-21 by jonekdahl