lukevp/ESC-POS-.NET #271: Intermittent Printing Issues on Cash Register Computers with Epson TM-T20III and ESCPOS_NET

Yalcinmehmet, 2024-04-20:
Environment:
- Operating System: Windows
- Application Framework: .NET 8.0, C#
- Library Version: ESCPOS_NET 3.0.0-
- Printer Model: Epson TM-T20III

`
private async Task PrintSale(Sale sale)
{
    try
    {
        string printerIPAddress = this.PrinterIPAddressTextBox.Text;
        IPAddress ipAddress;
        bool isValidIP = IPAddress.TryParse(printerIPAddress, out ipAddress);

        if (!isValidIP)
        {
            MessageBox.Show($"'{printerIPAddress}' is not a valid IP address.");
            return;
        }

        string hostName = ipAddress.ToString();
        Advertisement advertisement = this.GetRandomAdvertisementFromList();
        var printer = new ImmediateNetworkPrinter(new ImmediateNetworkPrinterSettings { ConnectionString = $"{hostName}:{port}", PrinterName = printerName });

        var epson = new EPSON();

        await printer.WriteAsync(epson.Initialize());
        await printer.WriteAsync(
          ByteSplicer.Combine(
            epson.CodePage(CodePage.WPC1254_TURKISH),
            epson.RightAlign(),
            advertisement?.LogoPath == null ? epson.Print("\n") : epson.PrintImage(File.ReadAllBytes(advertisement.LogoPath), false),
            epson.LeftAlign(),
            epson.SetStyles(PrintStyle.None),
            epson.SetStyles(PrintStyle.DoubleWidth | PrintStyle.DoubleHeight | PrintStyle.Bold),
            epson.LeftAlign(),
            epson.PrintLine($"{sale.Item.Name}"),
            epson.Print("\n"),
            epson.SetStyles(PrintStyle.None),
            epson.SetStyles(PrintStyle.Bold),
            epson.Print($"{sale.Item.Number} Ad. {sale.Item.Sum} Euro"),
            epson.Print("\n"),
            epson.SetStyles(PrintStyle.None),
            epson.Print("\n"),
            epson.PrintLine($" {sale.SaleTime.ToString("dd.MM.yyyy - HH:mm:ss")}"),
            epson.Print("\n"),
            epson.PartialCutAfterFeed(5)
          )
        );

        printer = null;
        await Task.Delay(1000);
    }
    catch (Exception ex)
    {
        MessageBox.Show(ex.Message);
    }
}`

Issue Description:
The application works as expected on my development machine, allowing me to print multiple receipts consecutively without issues. However, on the cash register computers, which are less powerful, prints are intermittently aborted. After a print job is aborted, the printer becomes unresponsive and cannot be accessed until it is restarted.

Yalcinmehmet, 2024-04-21:
I'm planning to change the implementation of our printing process to potentially solve the intermittent printing issues on the cash register computers. Instead of sending print commands consecutively via PrintAsync for each part of the receipt, I intend to gather all the data first into a single byte array. This array will then be sent to the printer in one go. This approach should help in reducing the overhead of multiple network transactions and may manage printer resources more efficiently. I'll update this thread with the results of this modification to see if it resolves the unresponsiveness and failure issues we are experiencing.

igorocampos, 2024-05-21:
Sounds like your solution is what I would suggest :) 

Hope it went well.