lukevp/ESC-POS-.NET #277: Printing issue with TM-T20III on 3.0 version

us259er, 2024-09-03:
Hi. I have an issue: recently I upgraded the library from version 1.6 to 3.0 and my printer suddenly stopped working. Even the code from readme does absolutely nothing. On older versions everything works fine... but I need new features like encoding selection.

igorocampos, 2024-09-18:
Hi @us259er are you able to debug and find what exactly is not working?

As far as needing encoding, that haven't change. The encoding is 100% optional, and if not informed, code will use the same as previous versions. You can check that out [here](https://github.com/lukevp/ESC-POS-.NET/blob/a211dfb22ae5b8fef2339ee0328fb62e981198fa/ESCPOS_NET/Emitters/BaseCommandEmitter/PrintCommands.cs#L18-L24)

us259er, 2024-10-31:
Hello, sorry for not replying for almost 2 months. I bought a second printer: TM-T88V to check if my previous one was faulty, but I couldn't print with this one either. I have resolved my issue by adding
```csharp
Thread.Sleep(500);
```
at the end of my program.

closed 2025-03-06 by us259er