lukevp/ESC-POS-.NET #279: SerialPrinter bug?

koo9, 2024-09-10:
it looks like the write(byte[] data) only put the byte[] into the queue but not sending it to the COM port. also 

what's line 171 - 181 do? it seems like the condition can be combined into one. 

`if (!IsConnected)
            {
                Logging.Logger?.LogInformation("[{Function}]:[{PrinterName}] Attempted to write but printer isn't connected. Attempting to reconnect...", $"{this}.{MethodBase.GetCurrentMethod().Name}", PrinterName);
                Reconnect();
            }

            if (!IsConnected)
            {
                Logging.Logger?.LogError("[{Function}]:[{PrinterName}] Unrecoverable connectivity error writing to printer.", $"{this}.{MethodBase.GetCurrentMethod().Name}", PrinterName);
                throw new IOException("Unrecoverable connectivity error writing to printer.");
            }`

igorocampos, 2024-09-11:
Hi, I took a look at Serial Printer and it doesn't have any specific implementation of Write method you mentioned, which means it uses the base class one, also used in all other printer classes. 

Dequeue from buffer doesn't happen in the Write method, but rather at `WriteLongRunningTask` which is a background task that does the actual writing to the binary writer (in your case it's the COM port, as you can check in the Serial Printer constructor)

My suggestion for your debugging is to put a break point [here](https://github.com/lukevp/ESC-POS-.NET/blob/a211dfb22ae5b8fef2339ee0328fb62e981198fa/ESCPOS_NET/Printers/BasePrinter.cs#L106) so you know bytes successfully dequeued from the buffer and are about to get written in the writer, and also [here](https://github.com/lukevp/ESC-POS-.NET/blob/a211dfb22ae5b8fef2339ee0328fb62e981198fa/ESCPOS_NET/Printers/BasePrinter.cs#L96) to know if IsConnected flag isn't `false` for whatever reason.

koo9, 2024-09-11:
@igorocampos yes, the SerialPrinter class  derive from the BasePrinter, the write(byte[] data) method only queue the data in ot the concurrentqueue but not writing it to the serial port. so it's actually not sending any data to the serial port.

igorocampos, 2024-09-11:
And has it? I'm not saying that there's no bug, but my gut says it's more likely to have a serial port connection/usage issue, rather than a code logic one, since the same code is working elsewhere.

koo9, 2024-09-12:
turned out that the printer support ESC/POS mode but not direct IP print, it needs to go through a serial port mapping to the printer. so closing this issue.

closed 2024-09-12 by koo9