lukevp/ESC-POS-.NET #284: Set print width

jasikpl123, 2024-10-28:
Hey, 
i would like to set print width by limiting to 40 characters per line (using font B). 
It work perfectly fine until I want to align to center - the printer just prints it thinking i'm using 80mm paper (i want to use 58mm instead) so the message is cut in half...
Any solutions?
Thanks x

igorocampos, 2024-10-28:
Which commands are you using to align text to center?

jasikpl123, 2024-10-28:
> Which commands are you using to align text to center?

e.CenterAlign()

igorocampos, 2024-10-28:
From EPSON's handbook:
[image]
In red I believe is the why you are getting your current behavior

jasikpl123, 2024-10-28:
Then how do i solve it? Can i specify the number of characters in one line when justifying the text?

igorocampos, 2024-10-28:
But if you know your line is always gonna be 40 chars long, it is probably easier to build an internal method that pads empty space in order to center align your text.

quick search in the internet, and found this for example
```cs
static string center(string text)
{
	int paperWidth = 40;
	int stringWidth = text.Length;
	int spaces = (paperWidth / 2) + (stringWidth / 2);

	return message.PadLeft(spaces);
}
```