lukevp/ESC-POS-.NET #285: Get information about the Pos printer model

jzeb, 2024-10-29:
May I ask whether I have obtained information about the POS printer model in the current LAN through network scanning

lukevp, 2024-10-29:
I don’t understand the question. This library implements a subset of the ESC/POS spec. There aren’t currently any methods implemented that give you model information, but if that’s part of the spec we would gladly accept PRs that add that functionality. If you’re asking about port / ip address scanning to find devices, that’s out of scope for this library and there should be many .net tools that can do this.

jzeb, 2024-10-30:
What I mean is that ESC-POS.NET can scan the LAN printer and obtain the corresponding brand and model of the printer

jzeb, 2024-10-30:
I really want to get my reply. Now I want to find a solution. Thank you

igorocampos, 2024-10-30:
Hi @jzeb, unfortunately, what you wish for doesn't have anything to do with the purpose of this library. You would need to accomplish that with another library and THEN use this one to communicate with the printer you found in your network

> What I mean is that ESC-POS.NET can scan the LAN printer and obtain the corresponding brand and model of the printer

> That’s out of scope for this library

jzeb, 2024-10-31:
> Hi @jzeb, unfortunately, what you wish for doesn't have anything to do with the purpose of this library. You would need to accomplish that with another library and THEN use this one to communicate [...]

So far, I have been looking for a long time, but I have not found a suitable library to use. Could you recommend it for me? Thank you

igorocampos, 2024-10-31:
Unfortunately I've never had that need, so I wouldn't know which to recommend, but quick search in Google pointed a few already, not sure if you already went that path or not.

jzeb, 2024-11-01:
Currently can not find a solution, in distress