lukevp/ESC-POS-.NET #288: Duplicate prints

butskristof, 2025-02-11:
Since #213 was closed, I want to bring up this issue again and look into why this is happening. 

For about half of the print commands I do in my application, the print happens twice. I'm not printing overly large tickets, and am sure that the `printer.Write()` is only executed once. 

I did upgrade the application from .NET 7 to .NET 9 and did some architectural changes, but I also created a minimal reproduction which behaves the same. The printer (TM-T20II) is connected directly over USB and accessed through a FilePrinter object. 
Some scenarios in which I noticed this: 
* application running on a Raspberry Pi in Docker (`/dev/usb/lp0` passed in as device)
* application running in a Linux VM (Debian) on Apple Silicon Mac in Docker (`/dev/usb/lp0` passed in as device)
* application running in a Linux VM (Debian) on x86 Intel machine on Docker (`/dev/usb/lp0` passed in as device)
* `dotnet run` in a Linux VM (Ubuntu) on x86 Intel machine directly, USB device passed through to VM 

A simple `echo "Hello" >> /dev/usb/lp0` always seems to yield only one print, but maybe I could try this with a more difficult print. 
Printing the test ticket with the configuration utility on Windows also always seems to do just one print. 

I created a very simple minimal application which creates a printer instance, prints the example from the readme and does `Thread.Sleep(500)` to give some time to print the buffer. Calling that three times in a row with a delay in between (to get the lock off the lp0 file) already results in duplicate prints. It's not deterministic which print will be duplicated however. 

Everything points to something in the library or a regression in a newer .NET version, but I can't quite fathom what it could be or how to track this down. I used practically the same code in earlier versions which behaved as expected. 
I'll give it another go without the `.PartialCut()` as well, as that seems to be the boundary after which it repeats itself. 

```
// Program.cs
using PrinterTest.Services;

var builder = WebApplication.CreateBuilder(args);

builder.Services
    .AddScoped<TicketPrintingService>();

var app = builder.Build();

app.MapGet("/", () => "Hello World!");

const int count = 3;
for (var i = 0; i < count; ++i) {
	using (var scope = app.Services.CreateScope())
	{
	    var service = scope.ServiceProvider
		.GetRequiredService<TicketPrintingService>();
	    service.Print();
	}
	Thread.Sleep(5000);
}

//app.Run();
```

```
// TicketPrintingService.cs
using ESCPOS_NET;
using ESCPOS_NET.Emitters;
using ESCPOS_NET.Utilities;

namespace PrinterTest.Services;

internal sealed class TicketPrintingService
{
    public void Print()
    {
        using var printer = new FilePrinter(filePath: "/dev/usb/lp0", createIfNotExists: false);
        var e = new EPSON();
        var commands = ByteSplicer.Combine(
	    e.CenterAlign(),
	    //e.PrintImage(File.ReadAllBytes("images/pd-logo-300.png"), true),
	    e.PrintLine(""),
	    e.SetBarcodeHeightInDots(360),
	    e.SetBarWidth(BarWidth.Default),
	    e.SetBarLabelPosition(BarLabelPrintPosition.None),
	    e.PrintBarcode(BarcodeType.ITF, "0123456789"),
	    e.PrintLine(""),
	    e.PrintLine("B&H PHOTO & VIDEO"),
	    e.PrintLine("420 NINTH AVE."),
	    e.PrintLine("NEW YORK, NY 10001"),
	    e.PrintLine("(212) 502-6380 - (800)947-9975"),
	    e.SetStyles(PrintStyle.Underline),
	    e.PrintLine("www.bhphotovideo.com"),
	    e.SetStyles(PrintStyle.None),
	    e.PrintLine(""),
	    e.LeftAlign(),
	    e.PrintLine("Order: 123456789        Date: 02/01/19"),
	    e.PrintLine(""),
	    e.PrintLine(""),
	    e.SetStyles(PrintStyle.FontB),
	    e.PrintLine("1   TRITON LOW-NOISE IN-LINE MICROPHONE PREAMP"),
	    e.PrintLine("    TRFETHEAD/FETHEAD                        89.95         89.95"),
	    e.PrintLine("----------------------------------------------------------------"),
	    e.RightAlign(),
	    e.PrintLine("SUBTOTAL         89.95"),
	    e.PrintLine("Total Order:         89.95"),
	    e.PrintLine("Total Payment:         89.95"),
	    e.PrintLine(""),
	    e.LeftAlign(),
	    e.SetStyles(PrintStyle.Bold | PrintStyle.FontB),
	    e.PrintLine("SOLD TO:                        SHIP TO:"),
	    e.SetStyles(PrintStyle.FontB),
	    e.PrintLine("  FIRSTN LASTNAME                 FIRSTN LASTNAME"),
	    e.PrintLine("  123 FAKE ST.                    123 FAKE ST."),
	    e.PrintLine("  DECATUR, IL 12345               DECATUR, IL 12345"),
	    e.PrintLine("  (123)456-7890                   (123)456-7890"),
	    e.PrintLine("  CUST: 87654321"),
	    e.PrintLine(""),
	    e.PrintLine(""),
	    e.PartialCut()
        );
        printer.Write(commands);
        Thread.Sleep(500);
    }
}
```

butskristof, 2025-02-11:
Some additional findings:
* same result on Debian 11 instead of 12 
* If I write to a normal file and then do `cat output >> /dev/usb/lp0`, that just works 
* .PartialCut() seems unrelated, it happens without cutting as well 
* it happens both when I keep the reference to FilePrinter in a singleton or whether I create a new one each time 
* dropping down the project to .NET 7 (and using the .NET 7 SDK) behaves the same

thegame4craft, 2025-07-18:
Same problem here. When i write my byte[] directly to the underlaying file (/dev/usb/lp0) sometimes duplicate prints come out.
I thinkerd some time and tried using a C# lock, but this doesnt solve the issue either

System Info Arch Linux (Kernel 6.12.36-1-lts)