lukevp/ESC-POS-.NET #296: QR Code Size

e0548965, 2025-06-12:
We are using this library and it works great, we print a QR code like this

` e.PrintQRCode("123456789", TwoDimensionCodeType.QRCODE_MODEL2, Size2DCode.EXTRA, CorrectionLevel2DCode.PERCENT_7),`

It works but despite having the largest setting the QR code still only takes up about 50% of the receipt width. We are using a standard 58mm thermal Bluetooth printer.

We would like to make the QR code as larger as possible, possibly spanning about 90% of the width of the paper or so in order to be able to scan it as efficient as possible. Is there a way to make the QR code larger than I already did above? If not, any tips to achieve this? I tried with images generated by QRCoder but just couldn't get that to work, the output I got was just random chars.

Thank you!

igorocampos, 2025-06-13:
Have you tried the other models of qr code? You mentioned using Model2, I would try the other options. If that doesn't cut it, then I believe you would have to generate a qr code image, and then print the image instead.

e0548965, 2025-06-14:
Thanks a lot for the reply @igorocampos been trying to do this but just cannot manage to get it right in a MAUI app, any reference/samples? I didn't find any. It just turns out totally gibberish chars and not sure what I am doing wrong. I generate the QR code with QRCoder.

e0548965, 2025-06-15:
This is my current code

```
byte[] pngBytes = GenerateQrImagePng("123456789");
await SavePngToDisk(pngBytes, "test-qr.png");
//I've verified here that test-qr.png is a 377x377 QR image
  
 var e = new EPSON();
 var payload = ByteSplicer.Combine(
   e.CenterAlign(),
   e.PrintLine(" "),
   e.PrintLine("--------------"),
   e.PrintImage(pngBytes, isHiDPI: true),
   e.PrintLine("--------------"),
   e.PrintLine(" ")
 );
```

And method to generate QR PNG

 ```
 public byte[] GenerateQrImagePng(string text, int targetWidth = 384)
  {
      using var qrGenerator = new QRCodeGenerator();
      using var qrCodeData = qrGenerator.CreateQrCode(text, QRCodeGenerator.ECCLevel.Q);
      var moduleCount = qrCodeData.ModuleMatrix.Count;

      int pixelsPerModule = targetWidth / moduleCount;
      int qrSize = pixelsPerModule * moduleCount;

      using var bitmap = new SKBitmap(qrSize, qrSize, true);
      using var canvas = new SKCanvas(bitmap);
      canvas.Clear(SKColors.White);

      using var paint = new SKPaint
      {
          Color = SKColors.Black,
          IsAntialias = false,
          Style = SKPaintStyle.Fill
      };

      for (int y = 0; y < moduleCount; y++)
      {
          for (int x = 0; x < moduleCount; x++)
          {
              if (qrCodeData.ModuleMatrix[y][x])
              {
                  canvas.DrawRect(x * pixelsPerModule, y * pixelsPerModule, pixelsPerModule, pixelsPerModule, paint);
              }
          }
      }

      using var image = SKImage.FromBitmap(bitmap);
      if (image == null)
          throw new Exception("SKImage.FromBitmap failed");

      using var data = image.Encode(SKEncodedImageFormat.Png, 100); // use PNG
      if (data == null)
          throw new Exception("SKImage.Encode returned null — PNG format not supported");

      return data.ToArray();
  }
```

When I pass this to PrintImage it just prints a lot of random chars and spacing, any tips?

closed 2025-06-15 by e0548965